rusefi/rusefi #5206: Reduce manual labor: yaml to auto-generate C header

rusefillc, 2023-03-29:
I know title sounds odd please hear me out.

Problem statement:
I am clearly still typing same thing twice:
first I open yaml and I do
```
  - pin: B23
# H144_OUT_PWM4
    id: C8
    class: outputs
    ts_name: --- VTC VVT
    function: VTC VVT
```

(not how I have to say ID = C8 for my custom java in rusEFI while leaving a human-readable comment H144_OUT_PWM4 which is hellen_meta.h)

and then I go edit boards.cpp file where I type
```
    engineConfiguration->vvtPins[0] = H144_OUT_PWM4;

```

Note how I am manually keeping things in sync?

No idea what exactly I am looking for but for instance

if I have machine-generated
```
#define B23_VTC_VVT C8
```

I would at least 
```
    engineConfiguration->vvtPins[0] = GPIO::B23_VTC_VVT ;
```

which is one step closer to not manually syncing yaml and C

chuckwagoncomputing, 2023-03-30:
Questions:
1. Do we want enum or preprocessor defines?
2. If answer to 1 is enum, we need a different name than `GPIO` as we already have `Gpio`
3. Is the value of the `ts_name` field the desired name of the field/variable?

rusefillc, 2023-04-04:
1. i am happy either way to be honest
2. either way it would probably have board-specific prefix like MICRORUSEFI_INJECTOR_1 or BoardMicroRusEFI::INJECTOR_1
3. yes