rusefi/rusefi #5211: 4g is symmetrical crank not just four stroke crank

rusefillc, 2023-03-31:
[image]

[image]

VVT sync needs to know if crank has single (60/2) or double (miata NB) or tipple (vg35) or else symmetrical crank

at the moment 2/0 custom toothed wheel is not smart enough to declare itself FOUR_STROKE_SYMMETRICAL_CRANK_SENSOR

is this what upsets @ElDominio ?

rusefillc, 2023-03-31:
New episode of #5111

@ElDominio please update all three boards to same latest snapshot

please use new "Mitsu 4G63 Crank" trigger

ElDominio, 2023-04-01:
[new trigger log car turned on.zip](https://github.com/rusefi/rusefi/files/11128548/new.trigger.log.car.turned.on.zip)

Cranked 4 times, engine started twice in the log.
Might be due to tuning, will test on April 1

ElDominio, 2023-04-01:
revB using new trigger:
[nonmodifiednewtrigger.zip](https://github.com/rusefi/rusefi/files/11130351/nonmodifiednewtrigger.zip)

revF with modified crank trigger
[modified ecu idle log.zip](https://github.com/rusefi/rusefi/files/11130352/modified.ecu.idle.log.zip)

rusefillc, 2023-04-01:
@ElDominio which of two modified revF? do we need cap mod or not?

ElDominio, 2023-04-01:
[image]

This one is working

closed 2023-04-02 by rusefillc