rusefi/rusefi #5214: Ignition/Fuel based soft RPM limit

ElDominio, 2023-04-02:
A correction factor added (x) RPM _before_ the hard cut RPM limiter would help make the limiter not-so harsh by substracting (y) degrees of timing and adding (z) % of fuel mass.  

(x) = window that the correction will be added throughout (example, if rpm limit is 7000, and (x) is 200, the corrections activate at 6800RPM, creating a 200rpm window)
(y) = degrees of timing REMOVED from actual timing during window
(z) = % of fuel ADDED during window

rusefillc, 2023-04-10:
@ElDominio please test we have https://github.com/rusefi/rusefi/pull/5228

ElDominio, 2023-04-11:
[image]

Does not work (timing should be 13 degrees in picture, and fuel time is unchanged)

ElDominio, 2023-04-27:
[image]

Problem
[image]

Just noticed: 
Soft limit timing is being appllied at hard limit hysterisis, meaning if RPM limit hysterisis is 500, the timing starts being retarded from 2000 instead of 1500
[image]