rusefi/rusefi #5224: RPM Limit should have hysteresis

ElDominio, 2023-04-07:
(x) = hysterisis
(Y) = RPM limit (hard limit where spark/fuel is cut)

if (y) is 7200rpm, and (x) is 200rpm, 
when the ECU sees 7200rpm, fuel/ign will cut, and stay cut until 7200-200=7000rpm is reached

rusefillc, 2023-04-07:
Part of #4529

rusefillc, 2023-04-07:
@ElDominio i believe we can reduce confusion with the following cheap proposal:

As you know on rusEFI online you can link to specific dialog like https://rusefi.com/online/view.php?msq=1438&dialog=cltFuelCorrCurve

i suggest we mention similar URLs of relevant existing settings and spell out existing setting name and maybe even suggest how many new parameters with what suggested name should be added?

ElDominio, 2023-04-07:
[image]

Nothing specific opens for me?

rusefillc, 2023-04-07:
Damn you are right some of these specific URLs are broken :( bug report https://github.com/rusefi/web_backend/issues/205

Try https://rusefi.com/online/view.php?msq=1438&dialog=ignitionSettings

ElDominio, 2023-04-08:
[image]

something like this

https://rusefi.com/online/view.php?msq=1438&dialog=limitsAndFallback

in here
[image]

rusefillc, 2023-04-10:
@ElDominio please test we have https://github.com/rusefi/rusefi/pull/5228

ElDominio, 2023-04-11:
[image]

This settings works
_________________________________

[image]

This has no rev limit

rusefillc, 2023-04-12:
Closed with defect

closed 2023-04-12 by rusefillc

andreika-git, 2023-04-14:
> This log shows not working rev limit at start, then change to NOT clt related and it worked fine with hysteresis

1) In the settings screen I see hard limit 7200, but in the log I see 3050 RPM max. How can rev limit work when not clt related?

2) Can you explain the inconsistency between the current value on the graph (>7000) and table (2500) on your screenshot?

[image]

ElDominio, 2023-04-15:
1) I switched the setting while logging

2) I have absolutely no idea? I didn't edit the screenshot or anything