rusefi/rusefi #5225: cheap build simplification?

rusefillc, 2023-04-07:
at the moment one cannot get a legit firmware by invoking just ``compile_proteus_f4.sh`` - one has to manually invoke ``gen_config_board.sh config/boards/proteus proteus_f4`` as well

that's not nice

would it be trivial just to move gen_config_board.sh invocation into common_make.sh to get a more resilient single-click local build?

rusefillc, 2023-04-07:
cheaper gap implementation of #4302

rusefillc, 2023-07-22:
@chuckwagoncomputing I would like to beg for your help here, are you in a position to volunteer any resources?

rusefillc, 2023-08-29:
https://stackoverflow.com/questions/5014632/how-can-i-parse-a-yaml-file-from-a-linux-shell-script

rusefillc, 2023-08-29:
Depends on https://github.com/rusefi/rusefi/issues/5546

rusefillc, 2023-10-24:
> cheaper gap implementation of #4302

@pchmura4 all this here is the general problem area

rusefillc, 2023-12-21:
https://github.com/rusefi/rusefi/issues/5817 has accidentally fixed this issue - ``compile_proteus_f4.sh`` produces a value firmware now (well, only if you do not need to actually change metadata)

A proper solution with automatic on-demand code generation is still needed

chuckwagoncomputing, 2024-03-08:
This is fixed, you can call firmware/bin/compile.sh and either pass it a meta-info file or pick one from a list.