rusefi/rusefi #5226: Latest build reads rawEGO, but not set to either Air Fuel Ratio gauge

ElDominio, 2023-04-07:
I can see rawEGO, but nothing on Lambda1 or 2
https://rusefi.com/online/view.php?log=1211

while uploading I also saw this on rusefi online
[image]

ElDominio, 2023-04-10:
If you check rawEGO channel, you will see voltage

If you check any Lambda or Air Fuel Ratio channel, you will see 0

rusefillc, 2023-04-10:
works on geehy discovery board

[image]

rusefillc, 2023-04-10:
[image]