rusefi/rusefi #5231: setSparkSkipRatio limitation

ElDominio, 2023-04-11:
> setSparkSkipRatio(0) to skip 0% of the ignition events, i.e. no skipping setSparkSkipRatio(0.5) would skip half of ignition events. We never skip two consecutive ignitions.

> **We never skip two consecutive ignitions.**

Why? If I want 95% skip ratio there probably has to be a few consecutive ignitions skipped.

At a certain point of skipping it should be allowed to cut consecutive ignitions, and at (1) it should fully disable ignition. Am I wrong to think this?
Was trying to use this function to write a custom two step code but no matter what number I used past (0.7), it would barely make a difference and keep increasing engine speed up to the rev limiter (since at least one channel had to fire if one was previously cut)

rusefillc, 2023-04-17:
I will add a separate setSparkHardSkipRatio function one day

ElDominio, 2023-05-25:
[sethardtoonly1.zip](https://github.com/rusefi/rusefi/files/11568052/sethardtoonly1.zip)
[ultrahardcut.zip](https://github.com/rusefi/rusefi/files/11568054/ultrahardcut.zip)

I set setSparkHardSkipRatio(0) and the engine still turned on

ElDominio, 2023-05-25:
[sparkskiphardtest.zip](https://github.com/rusefi/rusefi/files/11568116/sparkskiphardtest.zip)

you can see between gears ECU is bouncing hard skip between 0 and 1

ElDominio, 2023-05-25:
[image]

ElDominio, 2023-05-25:
Problem fixed! :D

closed 2023-05-25 by ElDominio