rusefi/rusefi #5235: We need to reshuffle timing calculation a bit, add one more gauge

rusefillc, 2023-04-12:
* we raw/base have timing we take from timing table
* we have per-cylinder actual timing all the way with per-cylinder correction

at the moment we have gauge for only base timing, and we've just added data points for per-cylinder

problem:
we need not-per-cylinder corrected timing "base + knockRetard + getLimitingTimingRetard" without per-cylinder

per-cylinder timing should stay

```
	angle_t sparkAngle =
		// Negate because timing *before* TDC, and we schedule *after* TDC
		- getEngineState()->timingAdvance[event->cylinderNumber]
		// Offset by this cylinder's position in the cycle
		+ getCylinderAngle(event->cylinderIndex, event->cylinderNumber)
		// Pull any extra timing for knock retard
		+ engine->module<KnockController>()->getKnockRetard()
		// Degrees of timing REMOVED from actual timing during soft RPM limit window 
		+ getLimpManager()->getLimitingTimingRetard();
```

ElDominio, 2023-04-12:
[tsetingsoftrevlimti.zip](https://github.com/rusefi/rusefi/files/11205982/tsetingsoftrevlimti.zip)

here is soft rev not really working

rusefillc, 2023-04-12:
totally introduced a defect (should be subtraction not addition) at https://github.com/rusefi/rusefi/commit/0470953cb8ab735237145229ecaf61897b868409 and no coverage has failed showing how weak coverage is :(

andreika-git, 2023-04-15:
We have a problem. We need to call LimpManager::onFastCallback() before EngineState::periodicFastCallback(), not after, because correctedIgnitionAdvance needs getLimpManager()->getLimitingTimingRetard() which uses LimpManager::updateState() which is updated from LimpManager::onFastCallback(). And we don't want to move the RPM limit logic to getLimitingTimingRetard() because both ignition and injection have common code.