rusefi/rusefi #5237: wow we have two separate per-cylinder trims?!

rusefillc, 2023-04-12:
we have per-cylinder table
```
angle_t getCylinderIgnitionTrim(size_t cylinderNumber, int rpm, float ignitionLoad) {
	return interpolate3d(
		config->ignTrims[cylinderNumber].table,
		config->ignTrimLoadBins, ignitionLoad,
		config->ignTrimRpmBins, rpm
	);
}
```

and we have
```
	// Plus or minus any adjustment if this is an odd-fire engine
	auto adjustment = engineConfiguration->timing_offset_cylinder[cylinderNumber];
```

rusefillc, 2023-04-12:
@mck1117 @ElDominio is this a feature or a bug?

mck1117, 2023-04-12:
Feature! The cylinder offset is for your crankshaft (odd fire engines, etc) and the per cylinder tables are for tuning.

ElDominio, 2023-04-12:
I used it for 180* angle 2 cylinder, can confirm

mck1117, 2023-04-12:
Mark uses it for Harley too. The timing mark is actually not lined up any cylinder, one is 22.5 ahead and the other 22.5 behind.

commit 588bf774 (2023-04-12): wow we have two separate per-cylinder trims?! fix #5237

refactoring: bringing two offsets much closer together, no change in overall logic
closed 2023-04-12 by rusefillc