rusefi/rusefi #5247: Staged injection support; use two sets of injectors on the same engine

ElDominio, 2023-04-21:
I'm going to be working on a rotary soon, and staged injection is extremely common in rotaries. 
 Staged injection uses primary injectors operating all the time and secondary injectors which operate only at higher load/rpm.  Both sets of injectors are controlled by the main fuel table, however the percentage of the total fuel delivered by each injector set is controlled by the Secondary Injection Staging Table.  The secondary injectors may be brought on at any stage after the lockout conditions are met by entering non-zero numbers into the Secondary Injection table.

There should be
A "percent injection" table, maybe 6x6, that decides how much injection is done by the second stage (0% meaning the primary set of injectors is used, 50% is both sets of injectors used to deliver the fuel equally (so, assuming 100mg of fuel is needed, the first set would inject 50mg, and so would the second), and 100% is all of the fuel being delivered by the secondary set.

This typically uses a clone of the primary injector settings (but now called "Secondary",) and the split ratio table.

[image]
X axis is RPM
Y axis is Load

ElDominio, 2023-08-12:
I'm starting to get into the bike market, and many 4 cylinder bikes have staged injector from the factory.

If it can't be done the previously described way, I could make it work with an RPM vs Load (tps or map) table that allows me to request injector duty/injection time for the extra set of injectors (without touching main fuel calculations)

ElDominio, 2023-10-14:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when a car has more than one set of injectors, but i can only control the first set.

**Would we need new configuration fields?**
Staged Injection Enable - Turns on the Staged Injection System
Staged Injection Type - There are many ways to control a seperate set of injectors working in tandem with a main one, and those options would be here.
_______________________________________________________________________
1. Simple - A mode where there is an additional tuning map for the second set of injectors. The injectors are controlled by their duty directly, without any fancy math about how much fuel is going through said set of injectors. I would **NOT** make this the default mode because of how hacky it is, but some people cannot comprehend how automatic staged injection works, so this will need to exist. This mode needs a table and pretty much nothing else. 
 _**1. The Table:**_  Load vs RPM table that relates directly to the duty cycle which the secondary set of injectors will be driven at. Literally that brain dead, since when they are not needed, they can be set to 0%.
__________________________________________________________________
2. Automatic - A mode where there is an additional map, but it is for deciding which set of injectors get used by %. For this, the injector data is **REQUIRED** for the second set of injectors, since they will be taken into account in the fuel model.

--Number of injectors - Number of injectors in the Staged Injection system; **can be different from the primary set of injectors**
--Staged Injector Flow Rate - Flow rate in cc/min of the Staged Injectors
--Staged Injector Dead Time - Same as the main ones, but for the Staged ones
--Staged Injector Fuel Pressure - MAP Reference, Fixed or Measured.
--Staged Injection % Split Table - This table tells the ECU which injectors fulfill what % of current fuel demand. If the engine is asking for 15mg of fuel, and the table is at 20%; the main injectors provide 80% of 15mg (so 12mg), and the staged injectors provide the remaining 20% (so 3mg). Table is Load vs RPM.

**Describe the solution you'd like**
I would like to run a second set of injectors synchronously with the engine.

**Describe alternatives you've considered**
Using an additional ECU talking through CAN to the main one, but this increases cost and complexity at time of install.

rusefillc, 2023-12-30:
older ask https://rusefi.com/forum/viewtopic.php?f=3&t=1846

JBR2110, 2023-12-31:
I really need that feature also! This would make the whole system 100% more user orientated.

rusefillc, 2023-12-31:
> I really need that feature also! This would make the whole system 100% more user orientated.

@JBR2110 noted. Please stop the harassment, at this point you are not helping but distracting.

commit 39d42220 (2024-01-08): Staged injection support; use two sets of injectors on the same engine fix #5247
commit 8f42f8cc (2024-01-08): Staged injection support; use two sets of injectors on the same engine fix #5247
closed 2024-01-08 by rusefillc