rusefi/rusefi #5248: Make trailing spark split a 3D table

ElDominio, 2023-04-21:
Split spark for rotaries should have a table to change the trailing angle over load vs RPM, 5x5 should suffice ( not much definition is needed since the main spark table is the main guide for spark timing.

[image]

ElDominio, 2024-02-23:
ooof this would be nice, 4x4 is enough but do we have space?

ElDominio, 2025-01-18:
@FDSoftware

FDSoftware, 2025-01-21:
-20 to +20 is ok for the values ​​of this table?, also is it also possible to use the table in the image as default value?

ElDominio, 2025-01-21:
The table shown is SUBSTRACTED from the main table, meaning trailing spark runs at 0 degrees if the main table is 12 and the value in the split table is also 12.

Is that understood?

PR #7394 (2025-02-07): Make trailing spark angle a 3D table #5248
* added a new table "trailingSparkAngle"
* removed trailingSparkAngle as config on rusefi.txt
* added trailingSparkAngle as variable on ignition state
* bump FLASH_DATA_VERSION

[image]

needs test/review from @ElDominio 

resolve: #5248
closed 2025-02-07 by rusefillc