rusefi/rusefi #5249: VVL activation

ElDominio, 2023-04-21:
The current way to activate VVL is using a Generic PWM table which is not very cool. 
This way makes the the VVL switch on and off when it activates and (maybe) falls a bit in RPM.

A simple option like this
[image]

Works, and add a TPS condition to not get it activating under a certain TPS or boost

rusefillc, 2023-04-22:
@ElDominio what is VVL? :) variable valve lift? on what cars?

ElDominio, 2023-04-22:
Yes, variable valve lift.

Toyota 2ZZ engine

ElDominio, 2023-07-24:
Also applies to K series.

Client had no idea how to use GP PWM table, this would greatly benefit them

Zazins, 2023-07-24:
This is basicly the same as I was thinking at #5407.
This is great way to do that as most ecu in market do it and for most of people easy to understand.

ElDominio, 2024-02-23:
Still relevant

ElDominio, 2024-05-25:
this should be tagged project-k

ElDominio, 2024-05-25:
this is also a clone, i will close and focus on #5704

closed 2024-05-25 by ElDominio