rusefi/rusefi #5251: Have different Error tiers

ElDominio, 2023-04-21:
There should be an "error" and a "critical error" tier.

Most errors could activate a rev limit at X RPM and "critical error" should works as is, but most errors should not be CRITICAL. Id-typical people write wrong numbers in many tables and then complain about the ecu being broken since they cannot read error messages

rusefillc, 2023-04-21:
@ElDominio please be more specific

We have two tiers "warning" and critical error

ElDominio, 2023-04-21:
Incorrect axis scaling "shouldn't" disable the ECU. Bad ETC calibration is arguably a good candidate for that. A table with bad scaling should either autocorrect to a "sane" value and throw a warning and maybe ignore the table until the scaling is fixed.

ElDominio, 2024-02-23:
This is still relvant, not really understood when will it throw a critical error and when only a warning