rusefi/rusefi #5253: Make MAP sync configurable

ElDominio, 2023-04-23:
A twin cylinder 300cc pulls 97kpa on cylinder one while cranking, but since the MAP threshold for MAP sync is not know or configurable, I believe it can't sync

This log -> https://rusefi.com/online/view.php?log=1222 has MAP sync disabled, we can see map periodically going from 99.9ishKPA to 97, which i believe to be the sync point. This log RUNS on WASTED SPARK AND BATCH FUEL
[image]

With MAP sync enabled, it never synchronizes and doesn't run.
This log-> https://rusefi.com/online/view.php?log=1223
Shows the same, but MAP sync was enabled and engine phase does not look correct; it DOES NOT RUN

[image]

rusefillc, 2023-04-23:
```
	if (engineConfiguration->vvtMode[0] == VVT_MAP_V_TWIN &&
			Sensor::getOrZero(SensorType::Rpm) < engineConfiguration->cranking.rpm) {
		// we are trying to figure out which 360 half of the total 720 degree cycle is which, so we compare those in 360 degree sense.
		auto toothAngle360 = currentPhase;
		while (toothAngle360 >= 360) {
			toothAngle360 -= 360;
		}

		if (mapCamPrevToothAngle < engineConfiguration->mapCamDetectionAnglePosition && toothAngle360 > engineConfiguration->mapCamDetectionAnglePosition) {
			// we are somewhere close to 'mapCamDetectionAnglePosition'

			// warning: hack hack hack
			float map = engine->outputChannels.instantMAPValue;

			// Compute diff against the last time we were here
			float diff = map - mapCamPrevCycleValue;
			mapCamPrevCycleValue = map;

			if (diff > 0) {
				mapVvt_map_peak++;
				int revolutionCounter = getTriggerCentral()->triggerState.getCrankSynchronizationCounter();
				mapVvt_MAP_AT_CYCLE_COUNT = revolutionCounter - prevChangeAtCycle;
				prevChangeAtCycle = revolutionCounter;

				hwHandleVvtCamSignal(TriggerValue::RISE, timestamp, /*index*/0);
				hwHandleVvtCamSignal(TriggerValue::FALL, timestamp, /*index*/0);
#if EFI_UNIT_TEST
				// hack? feature? existing unit test relies on VVT phase available right away
				// but current implementation which is based on periodicFastCallback would only make result available on NEXT tooth
				getLimpManager()->onFastCallback();
#endif // EFI_UNIT_TEST
			}

			mapVvt_MAP_AT_SPECIAL_POINT = map;
			mapVvt_MAP_AT_DIFF = diff;
		}
```

rusefillc, 2023-04-23:
@ElDominio it compares current with previous, no magic gap - all it cares is if above or below previous
``float diff = map - mapCamPrevCycleValue;``

Need to discuss this when we are both more available at the same time

rusefillc, 2023-04-23:
When @NMSTEC was touching Harkey Map cycle was waaaaay more visible. Is your engine an off fire? What is the angle?

ElDominio, 2023-04-23:
Odd fire 180
Cycle is 1-2-blank-blank

ElDominio, 2023-04-24:
Did the logs show the problem? This engine might NEED sequential because of high RPM + boost

NMSTEC, 2023-04-24:
Harley has extremely short runner (like 4 inches from valve to throttle), with an offset map sensor. 

Alpha, harley is running sequential.

rusefillc, 2023-04-24:
@ElDominio not great news: we have to agree the protocol for a ticket without a tune matching the log. Shall we start at $50 per ticket with a log link without tune link?

On more positive side now I am vaguely remembering relevant data point: it's not MAP it's instant MAP, and a few magic data points. Which tells me that it's not configured for Map sync maybe somehow? Which brings us to the fact that without proper tune link there is only that much I can do.

[image]

ElDominio, 2023-04-24:
[image]

If that tune doesn't have MAP sync enabled, enable it. Only difference

rusefillc, 2023-04-24:
@ElDominio cam mode not selected, user configuration issue

[image]

ElDominio, 2023-04-24:
So you need me to select it as active when that's the only difference? It is not selected because it doesn't run when i select it.

rusefillc, 2023-04-24:
better

[image]

ElDominio, 2023-04-24:
[2022-06-14_20.46.00.mlg.zip](https://github.com/rusefi/rusefi/files/11312116/2022-06-14_20.46.00.mlg.zip)

Yamaha FZ09 with working MAP sync