rusefi/rusefi #5264: I2C Baro problems

ElDominio, 2023-04-30:


rusefillc, 2023-04-30:
@ElDominio we have at least three issues here
* calibration UI should be disabled when we use I2C since I2C does not use calibration
* we have validity check
```
	if (kilopascal > 120 || kilopascal < 50) {
		return unexpected;
	}
```
and that check does not seem to be enough in case of (your) breakage
* open question what's wrong with your unit. Please try your 4chanG those have same baro. my 2chanE reads 99KPa
* i guess it should be possible to disable I2C baro as a workaround for your issue

rusefillc, 2023-04-30:
@ElDominio please confirm power cycle after change from analog baro into i2c baro?

ElDominio, 2023-04-30:
User is dumb please close

commit 689cd6da (2023-04-30): I2C Baro problems fix #5264
commit f963450e (2023-05-01): I2C Baro problems fix #5264
closed 2023-04-30 by ElDominio