rusefi/rusefi #5275: Ford Coyote VVT Decoder

ElDominio, 2023-05-13:
Here is an oscope log
[newerckplog.zip](https://github.com/rusefi/rusefi/files/11469970/newerckplog.zip)

Here is a picture of an oscope with a LinkECU
https://user-images.githubusercontent.com/5685108/237802589-19142f27-6f60-4c18-838a-408e6f7845d8.jpg

rusefillc, 2023-05-13:
[image]
[image]

two of them have two wires, two of them have three
[image]
[image]

rusefillc, 2023-05-13:
So we see lazy hall?

[image]

rusefillc, 2023-05-13:
> So we see lazy hall?
>

Our hw is different could be falling or rising or inverted in RE, we just have to try all combos and compare vvt gap ratio log with expected sequence

rusefillc, 2023-05-14:
Expected gap sequence see https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#ford-coyote-vvt

ElDominio, 2023-05-15:
No VVT reading, I'm going to see if I have wired Hall cam and not VR cam and to what pin with oscope.

Log: https://rusefi.com/online/view.php?log=1234

ElDominio, 2023-05-15:
Changed cable to hall input 3 and scoped[image]

No input on rusefi console trigger counters are 0

rusefillc, 2023-05-15:
@ElDominio @andreika-git I vaguely remember 8chan fab or BOM drama around hall pull-ups

was it https://github.com/rusefi/alphax-8chan/issues/41?

ElDominio, 2023-05-16:
So, i turned it on and it read fine, but after a while it does not read again. I think it might have vibrated off R125

ElDominio, 2023-06-30:
Tried new modified 8chan"

i could only see 31A producing anything useable, but still no reading
relevant tune: 
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/11913334/CurrentTune.zip)

relevant log
[novvtsadface.zip](https://github.com/rusefi/rusefi/files/11913347/novvtsadface.zip)

I see VVT syncgapratio going up, maybe i needed to flip edge? Did not try that

ElDominio, 2024-02-23:
Intake VVT reads! Only missing exhaust VVT test

ElDominio, 2024-05-25:
I am pretty sure I provided exhaust VVT logic logs, i am commenting to bump

rusefillc, 2024-08-31:
@ElDominio what's the status? shall we close?

ElDominio, 2024-09-06:
Position is decoded correctly, VVT on Gen2 works weirdly because of factory way of working. New ticket about strange VVT operation?

rusefillc, 2024-09-06:
>  strange VVT operation?

yes please

closed 2024-09-06 by rusefillc