rusefi/rusefi #5291: TunerStudio warnings in bundle 2023.05.28

Gorfest-n, 2023-05-29:
TunerStudio warnings in bundle 2023.05.28 alphax-2/8/8chan
[warnings .txt](https://github.com/rusefi/rusefi/files/11593724/warnings.txt)

[image]

rusefillc, 2023-06-01:
```
1 Errors:
Error: Error: CurveGraph yBin not a valid Constant., Problem at:
[mainController.ini]:[Line:3642]: 		yBins = gearBasedOpenLoopBoostAdder
```

rusefillc, 2023-06-04:
Three or four errors in .ini file fixed thank you for the report!

Current release TunerStudio 3.1.08 shows close to a hundred of similar warnings
``Warning: bit Constant ignBlends2_blendParameter, contains fewer options (26) that expected(32), padding remaining with "INVALID"``

but current beta version TunerStudio 3.1.08.16 already does not consider this an issue, let's just live as is and one TS has a release those warnings would be gone

closed 2023-06-04 by rusefillc