rusefi/rusefi #5303: Dashpot for return-to-idle from coasting

ElDominio, 2023-06-02:
This is a continuation of #3940 
I have written a decently-working implementation on https://gist.github.com/ElDominio/8385f846d1e0bcc8d2e2569ff62d45e9 on lines 101 to 113, uses a few global variables but this is basically what i'm looking for

rusefillc, 2023-06-08:
See also https://github.com/rusefi/rusefi/issues/4284 no idea how much relavant

andreika-git, 2023-06-08:
That's an example of a ticket we need to improve. Specifically, I have no idea what's the problem statement. And I have no actionable items. While this ticket is attempting to build on a previous one, but such incremental approach with long history is not working to me at the moment. My preference is for each ticket to be atomic self-contained, focused and isolated.

ElDominio, 2023-06-08:
Does the Lua code not provide what's needed?

There needs to be some return-to-idle logic for transitioning from Coasting IAC to idle IAC, which is used to keep revs slightly higher and provide a smooth transition to idle. I don't know how else to explain it, the code I wrote works decently but having it built in. 

[image]

I don't know how else to explain this except showing @rusefillc  in person whenever I see him

andreika-git, 2023-06-08:
What if we use iacByTpsTaper (which we already have) and just add a hold+decay to it? It will imitate a slow pedal release, the same as the dashpot correction does. The taper will act normally when the pedal is pressed, and will enter the hold+decay mode when the pedal is released (i.e. when tps < idlePidDeactivationTpsThreshold). We may want to add iacByTpsHoldTime and iacByTpsDecayTime to make it configurable.

rusefillc, 2023-06-08:
Some comments are on https://github.com/rusefi/rusefi/pull/4486 which was attempting to implement https://github.com/rusefi/rusefi/issues/3940

I was under the impression that https://github.com/rusefi/rusefi/pull/4486 was since rolled back but it does not seem to be the case

ElDominio, 2023-06-08:
> What if we use iacByTpsTaper (which we already have) and just add a hold+decay to it? It will imitate a slow pedal release, the same as the dashpot correction does. The taper will act normally when [...]

That seems exactly what I'm looking for; only that it needs to work in open loop mode too (i saw idle PID which in my mind implies closed loop)

ElDominio, 2023-06-13:
[image]

[dashpottest.zip](https://github.com/rusefi/rusefi/files/11739353/dashpottest.zip)

I see it transitioning from Coasting to Idle, and there is no smooth changeover from coasting IAC to idle IAC (coasting is 55, idle is 50)

ElDominio, 2023-06-13:
[dashpotworks.zip](https://github.com/rusefi/rusefi/files/11739577/dashpotworks.zip)

[CurrentTune.zip](https://github.com/rusefi/rusefi/files/11739578/CurrentTune.zip)

rusefillc, 2023-06-13:
@andreika-git says "And don't forget the third setting - extra air by tps"

andreika-git, 2023-06-14:
More info on testing the dashpot:
https://rusefi.com/forum/viewtopic.php?p=47791#p47791

andreika-git, 2023-06-14:
It seems we need tooltips for the new config fields. 
@ElDominio Can you help with the text for tooltips for each of 3 settings (and then I add them to config)?

ElDominio, 2023-06-15:
I will write later tonight when at laptop

ElDominio, 2023-06-15:
I suggest changing "Extra idle air when throttle pressed" to "Dashpot Initial IAC Adder" 
and remove "coasting to idle"; the fact that it is a dashpot implies that.

Dashpot Initial IAC Adder - IAC Value added when coasting and transitioning into idle. A typical value is 2%
Dashpot Hold Time  - How long initial IAC adder is held before starting to decay
Dashpot Decay Time - How long it takes to remove initial IAC adder to return to normal idle.

commit 5c7c8332 (2023-06-15): Dashpot for return-to-idle from coasting fix #5303
closed 2023-06-15 by rusefillc