rusefi/rusefi #5304: Sequential shift cut

ElDominio, 2023-06-02:
https://gist.github.com/ElDominio/8385f846d1e0bcc8d2e2569ff62d45e9
Lines 84-99

There is a CAN trigger which indicates an upshift, and when the ECU sees this "upshift" flag it does what the Lua code shows. This could be very useful in firmware (and I ran out of space in Lua)

rusefillc, 2023-06-13:
@ElDominio is this related to torque? is this a torque increase or torque decrease action? what kind of settings would that torque action take?

ElDominio, 2023-06-13:
Torque decrease
The idea is to lower torque so the transmission doesn't have to deal with torque+gear change.
It needs to know when to apply (digital or can input)
how much to retard the ignition by
how much to cut spark by (a percent of events)
a decay time after the input is turned off to transition to normal operating state (timing comes back slowly/spark is cut progressively less and less)

Seuential transmissions also have this using digital inputs (i say this because of some possible future user might need it in their sequential transmission car/bike)

andreika-git, 2023-07-10:
@ElDominio Can you please add more details on implementation?

1) Is this feature a part of TCU (and should be disabled if TCU is disabled) or it can be used/activated independently using a generic digital input?

2) What's the exact title of the new dialog? "Torque decrease"? "Sequential shift cut"? Something else?

3) Where exactly it should be added to the menu? For example, under "Advanced", after "Shift solenoids" or elsewhere?

4) Your LUA code also contains ETB control via setEtbAdd(), but you didn't list it in the settings above (only mentioned the ignition retard and spark cut ratio). Do we need also ETB control for this feature? If yes, do we need two different EtbAdd values for half-throttle and full-throttle - as in your code? And do we need a configurable or built-in/fixed (50%) threshold?

andreika-git, 2023-07-11:
5. What's the magic behind 
```
if id == 0x218 then
	if data[3] == 33 or data[3] == 50 or data[3] == 67 or data[3] == 84 or data[3] == 101 then
		shiftInProgress = true
```
Do you want us to hard-code these values into the firmware? Is it car-specific and/or should be configurable?

ElDominio, 2023-07-11:
I will try to update details tonight, point by point

ElDominio, 2023-07-11:
`
Is this feature a part of TCU (and should be disabled if TCU is disabled) or it can be used/activated independently using a generic digital input?`

Independently with a generic digital input (or CAN input)

`What's the exact title of the new dialog? "Torque decrease"? "Sequential shift cut"? Something else?`

Shift cut should be fine, since it can be used in sequential, manual and automatic transmissions

`Where exactly it should be added to the menu? For example, under "Advanced", after "Shift solenoids" or elsewhere?`

I say under advanced, just generally. 

`Your LUA code also contains ETB control via setEtbAdd(), but you didn't list it in the settings above (only mentioned the ignition retard and spark cut ratio). Do we need also ETB control for this feature? If yes, do we need two different EtbAdd values for half-throttle and full-throttle - as in your code? And do we need a configurable or built-in/fixed (50%) threshold?`

At least in my code, I have ETB modifiers since it also helps reduce torque. Using ETB should be optional, and the different values are since at WOT, you need significantly more TB reduction to affect torque than in cruising, where sometimes removing 5% TB changes torque significantly

`What's the magic behind *snip*`

This is vehicle specific, at least on this car, TCU spits out over CAN if it is in a  shift-in-progress state, so I added these as triggers to shift cut. This could be passed to TunerStudio as a digital input to trigger normally, inside the firmware

andreika-git, 2023-07-21:
> This is vehicle specific, at least on this car, TCU spits out over CAN if it is in a shift-in-progress state, so I added these as triggers to shift cut. This could be passed to TunerStudio as a digital input to trigger normally, inside the firmware

How could it be passed? Do you want to use LUA to implement your custom CAN protocol? If yes, then we need to implement #5307 first, perhaps?

ElDominio, 2024-02-23:
I did this in Lua, i use it for flat-shift. 
https://rusefi.com/forum/viewtopic.php?f=5&t=2670&p=48866#p48866