rusefi/rusefi #5308: Console "Install Drivers" is broken, keeps opening cmd windows

ElDominio, 2023-06-06:
After installing the drivers, console keeps opening CMD windows showing "requesting administrative privileges" and closing them, this keeps happening until console is killed in Task Manager

mi-hol, 2023-06-11:
@ElDominio is this the same issue as described in https://github.com/rusefi/rusefi/issues/5162 ?
In case it is, please close this issue as duplicate

ElDominio, 2023-06-11:
I don't think so, install is successful but it keeps opening windows without error

ElDominio, 2023-09-23:
@rusefillc  @rusefillc  @rusefillc

closed 2024-02-23 by ElDominio

rusefillc, 2024-02-23:
> This was fixed when more people started exploding

I am not aware of anyone fixing it unless Microsoft has changed something? weird.

ElDominio, 2024-02-23:
I saw a thread on Facebook about it, then the probalem stopped about a week afterwards