rusefi/rusefi #5311: What is the best time to ensureArrayIsAscending / validateConfig

abelom, 2023-06-06:
All tables and curves require ascending order of axis

At the moment we validate that all axis are ascending only on ecu boot

Problem: if a user types in a bad (not ascending) axis they would not know until reboot

Open question: would it be better or worse to validate on any configuration change?

abelom, 2023-06-06:
@ElDominio do you have an opinion?
@mck1117 do you have an opinion?

abelom, 2023-06-06:
See also #5310

ElDominio, 2023-06-07:
I would argue it would be better on config change, but not kill the ECU with it. It should throw a giant error on main page but not disable engine running

rusefillc, 2023-06-07:
> but not kill the ECU with it. It should throw a giant error on main page but not disable engine running
what you say is totally best but totally not possible to the best of my knowledge

we have three options
* silent nothing and potentially confused users
* FATAL right away every time you fat finger
* more effort, maybe better logic: validateConfig 10 seconds after fat finger change with a change FATAL

ElDominio, 2023-06-07:
Would a change to fatal make engine stop running? People (me) tune their cars while driving and that might be quite dangerous

rusefillc, 2023-06-07:
FATAL while car is driving is a bad experience I guess we should keep as is

closed 2023-06-07 by rusefillc