rusefi/rusefi #5318: add license as machine readable data

mi-hol, 2023-06-11:
readme and license.txt state rusEFI is licensed as "GNU GENERAL PUBLIC LICENSE,  Version 3, 29 June 2007"
but this information is not availble in GH machine readable meta data LICENSE.md.

I suggest to add  LICENSE.md at level of organisation to avoid duplication

rusefillc, 2023-06-11:
Why do we care? any relevant URLs related to github license contention?

mi-hol, 2023-06-12:
Some contributors (including myself) care to what they donate their time.
Many rusefi  sub-repos miss a license statement (i.e. https://github.com/rusefi/web_backend/blob/master/www/index.html)

mi-hol, 2023-06-12:
> any relevant URLs related to github license contention?

"For your repository to truly be open source, you'll need to license it so that others are free to use, change, and distribute the software." from [GH licensing](https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/licensing-a-repository)

rusefillc, 2023-06-12:
Too many separate things piled here. Title says machine readable. Later on we've transition to human time. I have no idea what to do here, closing.

closed 2023-06-12 by rusefillc

mi-hol, 2023-06-12:
I'll provide a PR containing just LICENSE.md for rusefi/web_backend as the proper fix.
Am really surprised about how fast this got closed :(

mi-hol, 2023-06-12:
PR created.
Note: with admin permissions the fastest approach to add a license is described [here](https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-license-to-a-repository)