rusefi/rusefi #5345: Map reading at high load is noisy

ElDominio, 2023-06-20:
[image]

log: 
[2023-06-19_18.09.26.mlg.zip](https://github.com/rusefi/rusefi/files/11794940/2023-06-19_18.09.26.mlg.zip)

Tune:
[CurrentTune.zip](https://github.com/rusefi/rusefi/files/11794942/CurrentTune.zip)

Can there be  some software smoothing applied to MAP after a certain load/RPM? That spikiness triggers boost limit and forces me to raise boost cut.

rusefillc, 2023-06-20:
@ElDominio I am hesitant to apply hacks until I comprehend root cause and the big picture

what is the data rate of this log? 

[image]
 

[image]

rusefillc, 2023-06-20:
@ElDominio do you have any logs at low RPM and max data rate? does not have to be this vehicle. I want to see instantMAP signal shape with more details

overall I wonder if RC filter is properly set, i.e. if hardware is in good place. that looks like 4chan @Qwerty-OFF

rusefillc, 2023-06-20:
@ElDominio any ideas why would MAP noise increase so much with a relatively small RPM change? are there any real life processes causing real pressure to get noisier?

have you tried changing map averaging window settings?

[image]

rusefillc, 2023-06-20:
[image]

mck1117, 2023-06-20:
oscilloscope the MAP sensor to see the true story

ElDominio, 2024-02-23:
I have learned how to leverage MAP averaging to make these problems mostly non-existent, i think it's okay to close