rusefi/rusefi #5353: batch injection

rusefillc, 2023-06-25:
[image]

[image]

rusefillc, 2023-06-25:
@ElDominio says

[image]

ElDominio, 2023-06-25:
[image]

[image]

mck1117, 2023-06-25:
lol that's neat

[image]

closed 2023-06-25 by rusefillc