rusefi/rusefi #5380: 4g93.md defaults comments

ElDominio, 2023-07-03:
```
    // default "true"
    engineConfiguration->isInjectionEnabled = false;
    // default "Single Coil"
    engineConfiguration->ignitionMode = IM_INDIVIDUAL_COILS;
    // default "dynamic"
    engineConfiguration->timingMode = TM_FIXED;
    // default 0.0
    engineConfiguration->globalTriggerAngleOffset = 90.0;
    // default "Inactive"
    engineConfiguration->vvtMode1 = VVT_FIRST_HALF;
    // default "false"
    engineConfiguration->invertCamVVTSignal = true;
    // default 0.0
    engineConfiguration->vvtOffsets1 = 310.0;
    // default "false"
    engineConfiguration->isPhaseSyncRequiredForIgnition = true;
    // default 0.0
    engineConfiguration->cylinderBankSelect1 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect2 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect3 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect4 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect5 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect6 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect7 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect8 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect9 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect10 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect11 = 1.0;
    // default 0.0
    engineConfiguration->cylinderBankSelect12 = 1.0;
    // default 4.0
    engineConfiguration->benchTestOnTime = 5.0;
    // default 3.0
    engineConfiguration->idleStepperReactionTime = 0.0;
    // default 200.0
    engineConfiguration->idleStepperTotalSteps = 0.0;
    // default "true"
    engineConfiguration->stepperForceParkingEveryRestart = false;
    // default 15.0
    engineConfiguration->etbIdleThrottleRange = 5.0;
    // default 15.0
    engineConfiguration->acIdleExtraOffset = 0.0;
    // default 50.0
    engineConfiguration->acIdleRpmBump = 200.0;
    // default 2.0
    engineConfiguration->iacByTpsTaper = 0.0;
    // default 0.5
    engineConfiguration->acDelay = 0.0;
    // default 0.0
    engineConfiguration->warningPeriod = 10.0;
    // default "false"
    engineConfiguration->is_enabled_spi_2 = true;
    // default 500.0
    engineConfiguration->vvtControlMinRpm = 550.0;
    // default 5.333333
    engineConfiguration->vbattDividerCoeff = 5.852941;
    // default 3.0
    engineConfiguration->vssFilterReciprocal = 0.0;
    // default 1000.0
    engineConfiguration->driveWheelRevPerKm = 500.0;
    // default 1.0
    engineConfiguration->finalGearRatio = 0.0;
    // default 3.0
    engineConfiguration->adcVcc = 3.29;
    // default "false"
    engineConfiguration->isSdCardEnabled = true;
    // default 300.0
    engineConfiguration->boostCutPressure = 0.0;
    // default 0.5
    engineConfiguration->primingDelay = 0.0;
    // default 400.0
    engineConfiguration->engineChartSize = 300.0;
    // default "Launch Button"
    engineConfiguration->launchActivationMode = CLUTCH_INPUT_LAUNCH;`
    
    ```

Timing mode is fixed? Should be Dynamic
ignitionMode  = wasted spark
globalTriggerAngleOffset  = -280
vvtMode1 = 4G63 (no vvt)
acIdleRpmBump = 50
vvtControlMinRpm = 1000
boostCutPressure = 200

ElDominio, 2024-02-23:
what is this? what was trying to be acheived?

rusefillc, 2024-02-23:
We were looking to embed a better 4g93 default

We have a couple of similar issues

closed 2025-03-01 by rusefillc