rusefi/rusefi #5385: Updater needs JRE11, JRE only exists up to 8?

ElDominio, 2023-07-04:
[image]

But

https://www.java.com/en/download/manual.jsp

[image]

rusefillc, 2023-07-05:
yeeeee, that's going to be a bit confusing :(

https://www.oracle.com/java/technologies/javase/jdk11-archive-downloads.html

ElDominio, 2023-07-05:
That download is enormous and people usually have garbage internet, even in the mainland. 

Could the updater be split off from console then? Have a dedicated updating application that runs on potatoes?

rusefillc, 2023-07-05:
try https://github.com/adoptium/temurin11-binaries/releases/download/jdk-11.0.19%2B7/OpenJDK11U-jre_x64_windows_hotspot_11.0.19_7.msi

rusefillc, 2023-07-05:
https://rusefi.com/s/j

closed 2023-07-05 by rusefillc