rusefi/rusefi #5401: 3A92 not syncing

ElDominio, 2023-07-10:
[image]

[TuneAngLogFor3A92.zip](https://github.com/rusefi/rusefi/files/12009131/TuneAngLogFor3A92.zip)

Why is there a trigger error? Can see RPM and can see cam, but no intake VVT position. Why?

ElDominio, 2023-07-10:
https://rusefi.com/online/view.php?msq=1503

Current MMitsubicha (my 3cyl) config

NOTE: 
Shop installer is installing in parallel (because CVT), Mitsubicha is plug and play (stock ecu removed)

ElDominio, 2023-07-10:
I have two logs for non-inverted and inverted cam input, i can see the vvtSyncGapRatio doesn't make sense in either

[logs for cam triggers.zip](https://github.com/rusefi/rusefi/files/12009306/logs.for.cam.triggers.zip)

ElDominio, 2023-07-10:
[Mitsu3a92inshop.zip](https://github.com/rusefi/rusefi/files/12009340/Mitsu3a92inshop.zip)

rusefillc, 2023-07-11:
a lot broken on

[image]

rusefillc, 2023-07-11:
Found a smoking gun

[image]

rusefillc, 2023-07-11:
@mck1117 you remember a while back we had an option for "fast sync on cranking" or something and we've killed it to be always ON? turning it OFF would have been so nice here.

mck1117, 2023-07-11:
why?

ElDominio, 2023-07-11:
[3a92triggerNoTPS.zip](https://github.com/rusefi/rusefi/files/12009547/3a92triggerNoTPS.zip)

tps disabled, no more tps error

rusefillc, 2023-07-11:
> why?

@mck1117 I believe that this car has fucked up crank signal. We produce RPM based on reasonable portion of the wheel and then that RPM stays visible for two long while we are in "too many events, trigger broken"

Well, maybe even without "fast trigger sync based on 90" we would have not seen a much better picture. Maybe just need to make trigger sync more resilient by adding more expected gaps into the sequence. I guess we would revisit once we find out what's fucked with this signal in piggy back mode.

mck1117, 2023-07-11:
so then just set "require full sync for firing" to true

rusefillc, 2023-07-11:
@ElDominio data rate went DOWN? that's a nasty log for you to share :(

but even that shows crazy instant RPM, maybe we need smarter code to explode car in case of instant RPM so uneven

[image]

ElDominio, 2023-07-11:
My personal Mitsubicha, starting and running without touching anything else
[mitsubicha running lgos.zip](https://github.com/rusefi/rusefi/files/12021578/mitsubicha.running.lgos.zip)

running tune

[Mitsubicha running tune.zip](https://github.com/rusefi/rusefi/files/12021582/Mitsubicha.running.tune.zip)

ElDominio, 2023-07-11:
Fake mitsubicha thing cranking log
[crankingfakemitsu.zip](https://github.com/rusefi/rusefi/files/12021586/crankingfakemitsu.zip)

fake mitsu tune (i copied mitsubicha tune and changed CAM input, connected to C7, have tried on E6 with no difference)
[fake mitsbuicha current tune.zip](https://github.com/rusefi/rusefi/files/12021588/fake.mitsbuicha.current.tune.zip)

rusefillc, 2023-07-11:
@ElDominio please upload happy tune / happy log to rusEFI online and I will mention at https://github.com/rusefi/rusefi/wiki/Case-Studies

ElDominio, 2023-07-11:
at https://rusefi.com/online/view.php?log=1257

ElDominio, 2023-07-11:
log at max data rate of fake mitsu
[crankingmaxdataratefakemitsu.zip](https://github.com/rusefi/rusefi/files/12021880/crankingmaxdataratefakemitsu.zip)

ElDominio, 2023-07-11:
[bothsignalscut.zip](https://github.com/rusefi/rusefi/files/12021912/bothsignalscut.zip)

both ckp and cmp to stock ECU were cut, no sharing of signals

rusefillc, 2023-07-12:
@ElDominio  @mck1117 I think the problem here is really cool: with current login and gap settings we only syn on half of crank revolutions the ones in "easier" compression strokes maybe? Compare with different vehicle same engine https://github.com/rusefi/rusefi/issues/5404#issuecomment-1632795508

On this customers vehicle the special wide gap does not happen half of the times, pictures are six revolutions of crank but on three the wide tooth is not detected as wire, gap ratio is too vague to distinguish it from other teeth

Is this a hardware issue or not? If it's NOT a hardware issue we would have to implement custom trigger logic so that we only sync half of the time :(

[image]

mck1117, 2023-07-12:
can we crank the engine with the spark plugs out?

ElDominio, 2023-07-12:
Will have that log tomorrow in about 20 hours

ElDominio, 2023-07-13:
[nosparkcrank.zip](https://github.com/rusefi/rusefi/files/12041658/nosparkcrank.zip)

barely any variation in instant and perfectly clean gap ratios like my own

ElDominio, 2023-07-13:
[triggeroklog.zip](https://github.com/rusefi/rusefi/files/12041795/triggeroklog.zip)
here trigger is ok and sparks fire in correct sequence with no loss of trigger or order

ElDominio, 2023-07-24:
The engine started by lying to the ECU.

I set as 6 cylinder wasted spark and it synced up and runs.

ElDominio, 2023-07-27:
Log for the fake Mitsubicha:
https://rusefi.com/online/view.php?log=1260

ElDominio, 2024-02-23:
This ran and was handed to the owner. Ok to close?

closed 2024-02-23 by rusefillc