rusefi/rusefi #5404: 3A92 he says happy but 20% instant RPM swing

rusefillc, 2023-07-11:
[image]

@ElDominio please _link_ happy tune
@ElDominio please link pictured log which you consider happy

This ticket is to investigate/document the 20% instant swing

ElDominio, 2023-07-12:
This seems to be of cranking, and I don't think it's impossible since a 3 cylinder that fires every 240 degrees (there will be a pause between compression events which starter should speed up and slow down)

Mitsubicha works perfectly fine cranking and running, I assume it is not pissing off decoder because low compression engine build.

rusefillc, 2023-07-12:
@ElDominio no urgency but can you please log with spark plugs removed and wide open throttle? I know I can technically do same with any other car just need help :(

rusefillc, 2023-07-12:
This being a 3 cylinder note how gap ratio sequence changes depending on what stroke is aligned with sync point

pictured are six crank revolutions - note how three times we have almost exactly gap of 3.00 and three times it's almost exactly gap ration 2.50