rusefi/rusefi #5407: Fan PWM function/ Configurable Outputs

Zazins, 2023-07-17:
Hello! 

Would be nice to add function for fan outputs as PWM and based on more than coolant temperature, like oil temperature and aux sensor values.
For now this could be done by separate general purpose PWM, but 4x of them fills very quickly.

Could be very good addition to add general purpose outputs with IF state like if Rpm> 600 PinX is active high/low.
This could be more configurable like MS do etc. 

Thanks!

rusefillc, 2023-07-17:
Looks related to https://github.com/rusefi/rusefi/issues/5326

Zazins, 2023-07-18:
[image]
[image]

I was thinking something like this. 
Some of the new cars use electric thermostat other kind on of valves and this is very powerful tool in ECU to simply got two conditions to active etc. 
About PWM fan, this kind of settings is simple and easy to use and can be improved with configurable axis. See attached pictures.  
This would be nice to implement.

Zazins, 2023-07-18:
Does this kind of features will be posible in near future? 
This would be very user friendly I think and easy to configurate outputs. PWM fan option would be great, modern engines use then in stock for Ac and cooling.

ElDominio, 2025-01-18:
@FDSoftware

rusefillc, 2025-10-07:
@madrat also says "limping / bad sensor duty cycle is needed" potentially separate issue since potentially we need limping mode for CLT sensor even with on/off fan