rusefi/rusefi #5413: Proteus f7 firmware compile fails, vw_b6.lua, 8000 bytes from string of 1073

biohazardav, 2023-07-18:
During compilation of Proteus f7 firmware, compiler stops, log output shows vw_b6.lua warning "strncpy truncated copying 8000 bytes from a string of length 10731 [-Wstringgop-truncation] 

**Hardware**
 - Board: PROTEUS_STM32F7
 - Engine info: EZ30

**To Reproduce**
Steps to reproduce the behavior:
download master, attempt to compile with proteus selected as board and CPU set to ARCH_STM32F7

**Expected behavior**
successful compilation of F7 proteus bin. 

**Additional context**
I attempted a couple workarounds such as not treating warnings as errors being disabled, which successfully compiled and board appeared to function but was unsuccessful at communication with console or tunerstudio. Appears the problem affects the pages sent from the board and the generated ini does not fix this.

mck1117, 2023-07-18:
did you run `./gen_config_board.sh` for your board first?

biohazardav, 2023-07-18:
I just attempted after your reply this was the output of gen_config_board.sh:

`cyrax@DESKTOP-JKUDFGM /cygdrive/c/rusefi/rusefi/firmware
$ ./gen_config_board.sh proteus
Reusing configuration cache.

BUILD SUCCESSFUL in 6s
19 actionable tasks: 5 executed, 14 up-to-date
Configuration cache entry reused.
This script reads rusefi_config.txt and produces firmware persistent configuration headers
the storage section of rusefiXXX.ini is updated as well
BOARD_DIR=proteus SHORT_BOARDNAME=
Generating signature for
cat: proteus/prepend.txt: No such file or directory
BOARD_SPECIFIC_URL=[] for [] from [proteus]
BOARD_SPECIFIC_URL=[https://rusefi.com/s/wiki]
Using COMMON_GEN_CONFIG [
 -readfile OUTPUTS_SECTION_FROM_FILE console/binary/generated/output_channels.ini  -readfile DATALOG_SECTION_FROM_FILE console/binary/generated/data_logs.ini  -readfile LIVE_DATA_MENU_FROM_FILE console/binary/generated/fancy_menu.ini  -readfile LIVE_DATA_PANELS_FROM_FILE console/binary/generated/fancy_content.ini  -readfile LIVE_DATA_GAUGES_FROM_FILE console/binary/generated/gauges.ini  -ts_destination tunerstudio  -triggerInputFolder ../unit_tests  -with_c_defines false  -field_lookup_file controllers/lua/generated/value_lookup_generated.cpp controllers/lua/generated/value_lookup_generated.md  -java_destination ../java_console/models/src/main/java/com/rusefi/config/generated/Fields.java  -initialize_to_zero false  -signature tunerstudio/generated/signature_.txt  -signature_destination controllers/generated/signature_.h  -ts_output_name generated/rusefi_.ini  -prepend integration/rusefi_config_trigger.txt  -prepend proteus/prepend.txt  -board proteus]
class com.rusefi.ConfigDefinition Invoked with [-enumInputFile, controllers/algo/engine_types.h, -enumInputFile, controllers/algo/rusefi_enums.h, -definition, integration/rusefi_config.txt, -firing_order, controllers/algo/firing_order.h, -prepend, console/binary/generated/total_live_data_generated.h, -prepend, integration/rusefi_config_shared.txt, -tool, gen_config.sh, -readfile, OUTPUTS_SECTION_FROM_FILE, console/binary/generated/output_channels.ini, -readfile, DATALOG_SECTION_FROM_FILE, console/binary/generated/data_logs.ini, -readfile, LIVE_DATA_MENU_FROM_FILE, console/binary/generated/fancy_menu.ini, -readfile, LIVE_DATA_PANELS_FROM_FILE, console/binary/generated/fancy_content.ini, -readfile, LIVE_DATA_GAUGES_FROM_FILE, console/binary/generated/gauges.ini, -ts_destination, tunerstudio, -triggerInputFolder, ../unit_tests, -with_c_defines, false, -field_lookup_file, controllers/lua/generated/value_lookup_generated.cpp, controllers/lua/generated/value_lookup_generated.md, -java_destination, ../java_console/models/src/main/java/com/rusefi/config/generated/Fields.java, -initialize_to_zero, false, -signature, tunerstudio/generated/signature_.txt, -signature_destination, controllers/generated/signature_.h, -ts_output_name, generated/rusefi_.ini, -prepend, integration/rusefi_config_trigger.txt, -prepend, proteus/prepend.txt, -board, proteus, -enumInputFile, controllers/algo/rusefi_hw_enums.h, -c_defines, controllers/generated/rusefi_generated.h, -c_destination, controllers/generated/engine_configuration_generated_structures.h]
Opening logs/gen_config_.log
Reading firing from controllers/algo/firing_order.h
                field = "Ignition Output 2",                       ignitionPins2, {isIgnitionEnabled == 1 && (ignitionMode != 0 && cylindersCount >= 2) && (ignitionMode !=2 || twoWireBatchIgnition == 1 || ((firingOrder == 2) || (firingOrder == 7) || (firingOrder == 9) || (firingOrder == 11) || (firingOrder == 12) || (firingOrder == 16) || (firingOrder == 19) || (firingOrder == 20) || (firingOrder == 23) || (firingOrder == 25) || (firingOrder == 29)))}
                field = "Ignition Output 3",                       ignitionPins3, {isIgnitionEnabled == 1 && (ignitionMode != 0 && cylindersCount >= 3) && (ignitionMode !=2 || twoWireBatchIgnition == 1 || ((firingOrder == 1) || (firingOrder == 3) || (firingOrder == 4) || (firingOrder == 5) || (firingOrder == 9) || (firingOrder == 13) || (firingOrder == 14) || (firingOrder == 15) || (firingOrder == 18) || (firingOrder == 20) || (firingOrder == 23) || (firingOrder == 25) || (firingOrder == 27) || (firingOrder == 28)))}
                field = "Ignition Output 4",                       ignitionPins4, {isIgnitionEnabled == 1 && (ignitionMode != 0 && cylindersCount >= 4) && (ignitionMode !=2 || twoWireBatchIgnition == 1 || ((firingOrder == 5) || (firingOrder == 7) || (firingOrder == 12) || (firingOrder == 14) || (firingOrder == 16) || (firingOrder == 17) || (firingOrder == 22) || (firingOrder == 23) || (firingOrder == 25) || (firingOrder == 26) || (firingOrder == 27) || (firingOrder == 31)))}
                field = "Ignition Output 5",                       ignitionPins5, {isIgnitionEnabled == 1 && (ignitionMode != 0 && cylindersCount >= 5) && (ignitionMode !=2 || twoWireBatchIgnition == 1 || ((firingOrder == 4) || (firingOrder == 12) || (firingOrder == 15) || (firingOrder == 18) || (firingOrder == 23) || (firingOrder == 26) || (firingOrder == 30) || (firingOrder == 31)))}
                field = "Ignition Output 6",                       ignitionPins6, {isIgnitionEnabled == 1 && (ignitionMode != 0 && cylindersCount >= 6) && (ignitionMode !=2 || twoWireBatchIgnition == 1 || ((firingOrder == 13) || (firingOrder == 22) || (firingOrder == 23) || (firingOrder == 29) || (firingOrder == 30)))}
[... cut ...]
        at java.base/java.nio.file.Files.newByteChannel(Files.java:380)
        at java.base/java.nio.file.Files.newByteChannel(Files.java:432)
        at java.base/java.nio.file.Files.readAllBytes(Files.java:3288)
        at com.rusefi.IoUtil2.getCrc32(IoUtil2.java:36)
        at com.rusefi.IoUtil2.getCrc32(IoUtil2.java:52)
        at com.rusefi.ConfigDefinition.doJob(ConfigDefinition.java:168)
        at com.rusefi.ConfigDefinition.main(ConfigDefinition.java:52)
SystemOut Hook done for logs/gen_config_.log
`

rusefillc, 2023-07-18:
Over the phone I was attempting to ask to open https://github.com/rusefi/rusefi/issues and search for "lazy"

rusefillc, 2023-07-18:
Duplicate of https://github.com/rusefi/rusefi/issues/5225 which has manual instructions

rusefillc, 2023-07-18:
Technically real question is "why would anyone need to compile while not contributing improvement or reporting issues"?

closed 2023-07-18 by rusefillc

biohazardav, 2023-07-18:
I don't normally run into many issues compiling and my contributions are little because my only modification is to the init_tps.cpp because the code does not play nice with the accelerator pedals i often use. I can't contribute to code because i am not a experienced coder. I dont think my input would be as helpful as others. 

I understand your frustration, and i will not call you again. I understand this stuff can consume more time especially when people dont go through the proper channels for help or information and I am trying to do what i can do for the 4 or so current boards im running before asking for help.

rusefillc, 2023-07-18:
@biohazardav please help by document why do you need a custom init_tps.cpp not on this ticket