rusefi/rusefi #5414: init_tps.cpp,  sensor redundancy an handling

biohazardav, 2023-07-19:
I noticed quite a ways back that there was a modification to the init_tps.cpp which adds fault handling for redundant sensors that read too similar to each other. If the voltages are too similar the accelerator pedal fault is set and lose the accelerator pedal input. 

This fault handling breaks all 4 of my DBW vehicles, 2 on proteus and 2 on MRE. 
Ive been removing the fault handling myself and recompiling the firmware for this small change which hasnt been a lot o trouble for me.  Recently i discovered i cant successfully compile the latest firmware for the proteus with STM32_F7.

The 08-14 subaru pedal im running has sensors that read exactly the same, which trips the firmware up everytime i take the pedal past approx 14%

I think this fault handling handling should be removed, I also think there should be some option in MRE's to not use redundant sensors because the analog pins are already limited on those boards and I cannot get the  aux reuse pins to function properly which limits the input even further.

mck1117, 2023-07-19:
> I also think there should be some option in MRE's to not use redundant sensors because the analog pins are already limited on those boards

pinout limitations are exactly why these safety checks _must_ exist, so that people don't start driving around with an uncontrollable ETB.

biohazardav, 2023-07-19:
Thats a ridiculous claim considering we already are talking about home made builds and the chances of a stuck throttle are no different than with a poor cable throttle.

biohazardav, 2023-07-19:
The insurance policy of redundancy in the throttle of a passenger vehicle makes complete sense, but not here.

rusefillc, 2023-07-19:
> The 08-14 subaru pedal im running has sensors that read exactly the same
Really, honestly genuine question: is that a 6 wire pedal sensor? can you flip GND and power on one of the sides of the sensor to get opposite sensors instead of same sensors?

biohazardav, 2023-07-19:
I tried that but they are contacless sensors which do not work with the polarity reversed. And even if you did that technically speaking once both of the voltages reached a crossing point of the dame value the fault would be thrown and someone would experience a intermittent loss of throttle control as it goes in and out of fault handling.

mck1117, 2023-07-19:
> technically speaking once both of the voltages reached a crossing point of the dame value the fault would be thrown

This is not true - please familiarize yourself with how it works. The check done at startup makes sure that your _configuration_ isn't too similar for the primary/secondary. The curves are allowed to cross over at some point, that's fine, so long as both TPS sensors resolve to close enough % at that point.

closed 2023-07-19 by biohazardav

biohazardav, 2023-07-19:
I see now there was a change made approx a month ago for brz pedal. I was using the june 4th release before this. The compilation i made of the nightly has this function. i did not see this before. i will try the pre compiled bin here soon to see if that semi-secret brz mode that was added works for me.

we should talk soon. when your free.