rusefi/rusefi #5423: Change afterstart enrich defaults

ElDominio, 2023-07-21:
Change default afterstart enrichment valoes to
1.05
5 seconds

1.2 and 10 seconds leads many people to believe their tune is wrong, leading them to compensate in incorrect places and the car running poorly after afterstart fades out

mck1117, 2023-07-21:
seriously who is looking at the lambda during the first 10 seconds?

why is your sensor hot within 10 seconds of start?

ElDominio, 2023-07-21:
tf do I know

but i have heard complaints about this from a few of the people that buy the ECU lol

In any case, why doe we want to run 20% extra fuel for 10 seconds under all circumstances (coolant temps etc) as a default? What is being helped by this?

mck1117, 2023-07-21:
Because that's a very conservative starting point. Too rich on startup isn't hurting anything, but too lean makes people complain when their engine stumbles until the intake valves warm up.

ElDominio, 2024-02-23:
Fair.