rusefi/rusefi #5431: Tach output critical error

Gorfest-n, 2023-07-22:
rusEFI master.2023.07.22.alphax-4chan.3805642889

If the tachometer output is set to "A4 - Tachometer", then we get a critical error. 
Previous builds don't have this error.
[image]

rusefillc, 2023-07-22:
See https://github.com/rusefi/alphax-4chan/issues/146

Володя, а поменялся ли у нас a-EN?

rusefillc, 2023-07-22:
@Gorfest-n у нас там распиновка новых плат немного поменялась :( Сейчас видимо сломанны и старые, и новые платы. Сейчас попробую починить немного :)

rusefillc, 2023-07-22:
```
void boardInitHardware() {
	alphaEn.initPin("a-EN", H144_OUT_IO3);
	alphaEn.setValue(1);
```

Qwerty-OFF, 2023-07-22:
Видимо для тахометра выбрана не правильная нога. @rusefillc когда-то будет пора что то сделать с группами пинов OUT_IOx и IOx чтобы они не путались? H144_OUT_IO3 в rev.h это тестовая точка P19.
[image]

commit f6804823 (2023-07-23): fix https://github.com/rusefi/alphax-4chan/issues/146
fix #5431
commit 162507ca (2023-07-23): fix https://github.com/rusefi/alphax-4chan/issues/146
fix #5431
closed 2023-07-23 by rusefillc