rusefi/rusefi #5443: Document hardware setup for openocd_wipe_and_flash proteus

rusefillc, 2023-07-23:
cannot find my tc2030 stlink v3 cable and failing to use
```
.github/workflows/hw-ci/openocd_wipe_and_flash.sh ../.github/workflows/openocd_ci_proteus_f4.cfg
```
and
```
.github/workflows/hw-ci/openocd_wipe_and_flash.sh ../.github/workflows/openocd_ci_proteus_f7.cfg
```
using eBay stlink v2

We need an explicit comment right inside those files to document
* if USB or +12v power is required on top of st-link
* if v2 or v3 tc2030 or v3 jtag is required

rusefillc, 2023-07-24:
@dron0gus any idea why is v2 and v3 not interchangable?

mck1117, 2023-07-24:
those hw ci scripts contain an exact serial number of the stlink

rusefillc, 2023-07-24:
> those hw ci scripts contain an exact serial number of the stlink

yes some scripts contain exact serial but that's not my issue my f7 script does not contain exact serial
https://github.com/rusefi/rusefi/blob/master/.github/workflows/openocd_ci_proteus_f7.cfg

closed 2023-07-26 by rusefillc