rusefi/rusefi #5465: EGO sensor heater output inverted

Zazins, 2023-07-25:
Hi! 

Would be good and maybe easy feature to include is EGO sensor heater output to be inverted. Mainly this output is used to trigger VNLD etc. driver for sensor ground or PCB like Spartan OEM and more. In general this setting sets pin high when engine is off running but low when engine is running.

And could be possible same for indicator led outputs. 

Thanks!

rusefillc, 2023-07-25:
Lua?

rusefillc, 2023-07-25:
@Zazins I am confused why interested what is "high" and "low" - those terms are vague between logic level "high" and electrical which relates to specific hardware. Please elaborate why and what is needed, what existing stuff have you tried?

mck1117, 2023-07-25:
what behavior do you expect?

the current behavior is:
- when heating is desired, MCU pin goes to 3.3v
- when heating is not desired, MCU pin goes to 0v

Zazins, 2023-07-25:
I'm sorry about this. 
Thanks for information @mck1117, just double checked Proteus and it seems the VNLD is problem and the logic isn't the case. 
VNLD is controlling Spartan 2 OEM ground wire to control and protect sensor before engine running and it was all the time on. 
Just  I confused why it worked as it should when set as ignition output and  bench tested.
@rusefillc   Excuse me if I wasn't clear with correct terms, I meant before logic output like injectors has "Default" "Default Inverted' etc. 

Again sorry about my mistake.

rusefillc, 2023-07-25:
Not firmware issue. Please post on forum or https://github.com/mck1117/proteus if you suspect hardware design issue.

closed 2023-07-25 by rusefillc