rusefi/rusefi #5471: Setting Trigger to custom tooth with 1+0 on camshaft causes hang up on 12v

ElDominio, 2023-07-28:
**Describe the bug**
As stated in the title, when you set trigger settings to
custom toothed wheel
1 tooth
0 missing

and apply 12v, th ECU hangs until reset

**Hardware**
 - revD 2chan
 - Bike engine

**To Reproduce**
Steps to reproduce the behavior:
Apply settings as stated in first section
reset ECU
power up with USB
apply 12v to main 12v pin

**Expected behavior**
ECU works as normal

rusefillc, 2023-07-29:
interesting, unable to reproduce on a Proteus

[image]

rusefillc, 2023-08-02:
@ElDominio please provide two tunes "tune right before change" and "tune with change"

ElDominio, 2023-08-05:
Four cylinder ECU used in parallel to control injection

commit f3a5586e (2023-08-05): Custom Trigger Tooth with 1+0 causes endless loop fix #5471
closed 2023-08-05 by rusefillc