rusefi/rusefi #5492: proteus shows crazy RPM on Honda Generator

rusefillc, 2023-08-05:
https://rusefi.com/online/view.php?msq=1526

https://rusefi.com/online/view.php?log=1266

[image]

rusefillc, 2023-08-05:
@ElDominio says shunt resistor

http://www.useasydocs.com/theory/vr.htm

rusefillc, 2023-08-05:
[image]

rusefillc, 2023-08-06:
VR has one side ground on this "Vehicle"

VR and injector

[image]

rusefillc, 2023-08-06:
MRE "crank" and raw VR, with 10K pull-down

[image]

Qwerty-OFF, 2023-08-06:
Try 1k - 4.7k R1207, 10n C1209

More correctly, it seems to need to change MAX9924 operating mode, see Figure 1 and Table 1 in MAX9924 datasheet.

rusefillc, 2023-08-06:
@Qwerty-OFF do we have similar trick for tle8888?

rusefillc, 2023-08-08:
Aborting since no plans to test further :(

closed 2023-08-08 by rusefillc