rusefi/rusefi #5503: MAP Sync not working

ElDominio, 2023-08-09:
**Describe the bug**
Activated MAP sync and the engine does not turn on.

**Hardware**
RevF

**To Reproduce**
Set cam trigger to VTwin by MAP
Set MAP readout angle to 150
engine does not turn on

**Expected behavior**
MAP sync is achieved and engine runs, or at least, does not sync but still runs semi-sequentially

tune and log:
https://rusefi.com/online/view.php?log=1271

rusefillc, 2023-08-09:
[image]

rusefillc, 2023-08-09:
@ElDominio weird that MAP data point (red line) has stopped moving while rawMap is alive

@ElDominio looks like this log is 15 minutes since ECU boot and something is already stuck. Please reset ECU and provide a log of first cranking attempt.

rusefillc, 2023-08-09:
@ElDominio I've added some extra data points please grab a fresh log with latest firmware

ElDominio, 2024-02-23:
Will have to test soon, commenting to bump it. Have test vehicle in the shop