rusefi/rusefi #5511: EPIC: we need to test fresh boards

andreika-git, 2023-08-14:
Test usecase:
- all actuators (lowside) + all sensors for hellen125honda board
  * test used IO first, then figure out how to test unused IO
We have a special Proteus config which blinks all outputs, but this is a bad way, we need better.
Stim firmware already has console, and it knows how to test outputs.
Stim hardware already has 8 (or 10?) different voltages suitable for testing analog sensor inputs.
We need an instruction on pin connection map: which pin of stim

andreika-git, 2023-08-14:
Related issues:
~~https://github.com/rusefi/rusefi/issues/5436~~
https://github.com/rusefi/rusefi/issues/5437
https://github.com/rusefi/rusefi-hardware/issues/222

rusefillc, 2023-08-16:
See also https://github.com/rusefi/rusefi/issues/5514

rusefillc, 2023-08-19:
Actual board profiles https://github.com/rusefi/rusefi-hardware/issues/228

andreika-git, 2023-08-19:
Now we have https://github.com/rusefi/rusefi-hardware/commit/72c7d06cec56a038330b7406647478b9ae76a516
We can perform the first test of CAN communication...

rusefillc, 2023-08-21:
rusEFI gets CAN from stim
rusEFI might be not sending data correctly https://github.com/rusefi/rusefi/issues/5525

rusefillc, 2023-08-21:
#4630

andreika-git, 2023-08-21:
https://github.com/rusefi/rusefi-hardware/commit/32495af8984dca49b91e472f044ff7822e0a965e

@rusefillc Can you please test the raw_analog_values packet processing?

rusefillc, 2023-08-22:
> @rusefillc Can you please test the raw_analog_values packet processing?

rusEFI side was broken - we were relying on outputChannels which are not populated if USB is not connected. this is now fixed https://github.com/rusefi/rusefi/commit/9adc323fec207f9e49fc2725dd4ce8826075c42f

rusefillc, 2023-08-22:
> 
> @rusefillc Can you please test the raw_analog_values packet processing?

https://github.com/rusefi/rusefi-hardware/issues/228#issuecomment-1687371773

rusefillc, 2023-08-24:
```
Processing BENCH_TEST_BUTTON_COUNTERS ID=770004/l=8 19 0 0 0 0 0 0 0
Processing BENCH_TEST_EVENT_COUNTERS ID=770000/l=8 3E 22 22 1F 0 0 8C 0
```

pretty expected result for the following configuration (things are wires in the most reasonable way)

```
void proteusStimQc() {
    engineConfiguration->trigger.type = trigger_type_e::TT_ONE_PLUS_ONE;

   	engineConfiguration->triggerInputPins[0] = PROTEUS_DIGITAL_1;
   	engineConfiguration->triggerInputPins[1] = PROTEUS_DIGITAL_2;
   	engineConfiguration->camInputs[0] = PROTEUS_DIGITAL_3;
   	engineConfiguration->camInputs[1] = PROTEUS_DIGITAL_4;
   	engineConfiguration->vehicleSpeedSensorInputPin = PROTEUS_DIGITAL_5;
   	engineConfiguration->brakePedalPin = PROTEUS_DIGITAL_6;
}
```

rusefillc, 2023-09-17:
Good enough, done!

closed 2023-09-17 by rusefillc