rusefi/rusefi #5513: Renault F3R trigger

rusefillc, 2023-08-15:
60-2-2

closed 2023-08-15 by rusefillc

smclt30p, 2023-12-09:
That waveform capture is not good, that is with inversed polarity! The "spike" needs to be below the zero point. 

[image]

Also the trigger edge must be falling, as it's the faster edge and won't cause ignition drift.

rusefillc, 2023-12-09:
> The solution is to ignore the first missing gap completely and just trigger on the secondary missing gap.

Yes rusefi uses onlu rising fronts for VR

smclt30p, 2023-12-09:
That spike is there so that Renault ECU's can **detect** reverse polarity wiring and throw a DTC, because a standard service item on Renaults is to re-pin the crank sensor.

[image]

There is also half-a-dosen different connector types and connector keys between VR sensor manufacturers.

commit ee83fea7 (2023-08-15): Renault F 60-2-2 trigger wheel fix #5513