rusefi/rusefi #5539: EGO sensor type dropdown is broken, all settings the same

ElDominio, 2023-08-24:
**Describe the bug**
No matter which you choose, all use 0v=10afr, 5v = 20afr (AEM)

**Hardware**
rusefi firmware

**To Reproduce**
Go to sensors
pick ego sensor
choose valid input channel
go to "type" 
choose anything

**Expected behavior**
values change according to sensor type

**Actual behavior**
value does not change

andreika-git, 2023-08-28:
Did you try rebooting the ECU after saving the sensor type?

rusefillc, 2023-08-28:
I've made some progress but commit(s) are not linked to this ticket due to human error https://github.com/rusefi/rusefi/commit/aa713e4768a7fd475642f26295caca3530a3c60f

this is not finished - looks like "type" configuration property should be removed. narrow band seems to be a special case which maybe has never worked? open question if remove narrow band special case or detect it based on hard-coded magic values or else

ElDominio, 2024-04-06:
I think this was fixed?