rusefi/rusefi #5541: 4G69 Cam pattern

ElDominio, 2023-08-25:
**Is your feature request related to a problem? Please describe.**
We have no support for this cam, seems to be 4+1+1
4 teeth at 90 degrees, two of them have two teeth.

**Would we need new configuration fields?**
Nope, only dropdown choice for 4g69 cam

**Describe the solution you'd like**
Add support for 4G69 cam

**Describe alternatives you've considered**
grinding teeth but that's ugly

**Additional context**
[image]

rusefillc, 2023-08-25:
[image]

rusefillc, 2023-08-25:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#4g69-vvt

commit 6103b4f7 (2023-08-25): Mitsubishi 4G69 Cam pattern fix #5541
closed 2023-08-25 by rusefillc