rusefi/rusefi #5542: Injector duty reading half of reality

ElDominio, 2023-08-25:
**Describe the bug**
Injector duty is half of real, was hitting Fuel Cut Code 10 (duty) and injector duty was at 50%

**Hardware**
4chan revH

**To Reproduce**
rev to a place where you ask for more than 96% duty
check reported duty

**Expected behavior**
read 100% instead of 50%

[test trigeer.zip](https://github.com/rusefi/rusefi/files/12435070/test.trigeer.zip)

Log shows fuel cut code and current duty being 50%

rusefillc, 2023-08-25:
@ElDominio without tune I do not know if you are in GDI mode

with GDI mode setting we are limited to 50% duty cycle

ElDominio, 2023-08-25:
Can guarantee not GDi, but cannot guarantee not GDi mode; i wish this persons internet was not absolutely horrifying

andreika-git, 2023-08-28:
Let him check the "Number of lobes" setting (should be 0) in the Controller->"HPFP Cam Configuration" dialog.

rusefillc, 2023-08-29:
Seems to be caused by #5416

Default tune has three lobes and too many users are now affected. Changing check to look at output pin not lobes.

closed 2023-08-29 by rusefillc

rusefillc, 2023-08-29:
@ElDominio a work-around would be to set Controller->"HPFP Cam Configuration" -> "Number of lobes" to zero

ElDominio, 2023-09-19:
Was set to 3, the seller FINALLY gave me the tune

rusefillc, 2023-09-19:
@ElDominio that's rusEFI issue we had bad default and deciding GDI mode because of it. Gdi mode logic changed now.

commit 55cb5cfc (2023-08-29): Injector duty reading half of reality fix #5542