rusefi/rusefi #5559: Injector DutyCycle jumps to 0 when duty goes beyond 127.5%

tmbryhn, 2023-09-10:
**Describe the bug**
DutyCycle variable starts over from 0 after passing 127.5%.

**Hardware**
 - Custom hardware running latest Proteus_F4 release
 - Stimulator, simulating a V12 setup.

**To Reproduce**
1. Define all cells in VE table to a constant high value (I used 100).
2. Increase RPM steadily and observe Injector DutyCycle%.
3. Keep increasing RPM. Note Injector DutyCycle re-sets to 0 when duty goes beyond 127.5%, and keeps incrementing from 0 again as RPM keeps increasing.

**Expected behavior**
Injector DutyCycle% should not display values higher than 100%.
IF Injector DutyCycle% is to display values >100, the variable should increase monotonically with RPM, given that all other conditions/variables remaining constant, and not start over from 0 when a value of 127.5 is surpassed.

**Screenshots**
[image]

rusefillc, 2023-09-10:
@tmbryhn appreciate your effort to properly document the issue!

obviously I agree that actual duty cycle never exceeds 100% but here we log "commanded duty cycle" and there is value in logging that commanded is unrealistic

my options here are either to allocate 2 bytes for this data point or to reduce precision or to cap the logged value. I've chosen to just cap the logged value at 127 - so, commanded duty cycle 180% would be logged at 127%.

closed 2023-09-10 by rusefillc

tmbryhn, 2023-09-11:
Sounds like a good trade-off. Thank you.

commit aa00a9ec (2023-09-10): Injector DutyCycle jumps to 0 when duty goes beyond 127.5% fix #5559