rusefi/rusefi #5565: Fan not be able to be set because in use at VR PWM

ElDominio, 2023-09-15:
[image]

what's VR PWM?

ElDominio, 2023-09-16:
[image]

https://rusefi.com/online/view.php?msq=1541

rusefillc, 2023-09-16:
* sounds like 2chan never had two discrete VRs so we shall ignore the second threshold setting and hide the second channel control?
* sounds like 2chan before rev G should have hard-coded OUT_PWM6: VR threshold and not even expose the setting?
* sounds like 2chan rev G no longer needs VR threshold settings we should hide for that specific revision?

@ElDominio please confirm

ElDominio, 2023-09-16:
I have more dumb revision ideas for rev-after-G so I think they shouldn't be hidden

Having no second channel seems fine though

rusefillc, 2023-09-16:
> I have more dumb revision ideas for rev-after-G so I think they shouldn't be hidden

we will deal with latest once we are in the future :)

ElDominio, 2023-09-16:
I disabled PC8 and got the same error btw

rusefillc, 2023-09-17:
@ElDominio please update firmware to get more detailed message

[image]

rusefillc, 2023-09-17:
@ElDominio also VR pin controls are now hidden for 2chan and 4chan with, and 2chan values are hard-coded

[image]

ElDominio, 2023-09-18:
Problem is fixed now? Updated to 20230818 and all worked

commit 2f907af1 (2023-09-16): Fan not be able to be set because in use at VR PWM fix #5565
only:reducing confusion
closed 2023-09-18 by rusefillc