rusefi/rusefi #5566: 36/2/2/2 is confusing

rusefillc, 2023-09-16:
36/2/2/2 is confusing due to rotary usage (by RX-8?)

* why do allow cam placement? is that again about rotary use-case?
[image]

* image is not right
[image]

rusefillc, 2023-09-16:
Also TDC positions see https://www.rusefi.com/forum/viewtopic.php?p=48336#p48336

ElDominio, 2023-09-16:
Subaru and Mazda (non-rotary) use this trigger. I have seen Renesis engines with 36-2-2-2 turn on on RusEFI; what's the issue?

rusefillc, 2023-09-17:
Real log with sync having issues https://rusefi.com/online/view.php?log=1287