rusefi/rusefi #5569: ETB activates radiator fan

rusefillc, 2023-09-18:
https://rusefi.com/online/view.php?msq=1543

4chan, kia adapter, brand new patch cord

sitting in parked vehicle

* radiator fan bench test works, TS indicator goes green while bench test
* with "disable ETB if engine is stopped" FAN is not spinning, no TS gauge
* i chance "disable ETB if engine is stopped" to "false" to test ETB and I get fan spinning while TS indicator is not active

rusefillc, 2023-09-18:
second example of 4chan does same so probably not a fab defect

@ElDominio have you ever noticed anything similar?

ElDominio, 2023-09-18:
Maybe happens on new revision? I always leave ETB active with engine off and have never seen this behavior

rusefillc, 2023-09-25:
patchcord issue main relay wires was not hooked up properly

closed 2023-09-25 by rusefillc