rusefi/rusefi #5574: STM32H7 problem

beeedee, 2023-09-20:
I had two PCBs produced from the relevant PCB manufacturer, both of which were designed with Proteus v0.7. The only modification I made was to produce them with the STM32H723ZGT6 microcontroller. Now, I'm experiencing some issues while trying to download and install the Proteus Black Market (STM32H7) bundle. When I attempt to load it using the rusefi console.exe as I've seen in online tutorial videos, I can't seem to find the manual DFU option. Pressing the boot button during startup also doesn't seem to make a difference.

beeedee, 2023-09-21:
[image]

I think the visual can express me better

rusefillc, 2023-09-21:
one of two things
* you have wrong (OLD) DFU driver and you would have to update DFU driver
* or rusEFI console does not recognize that correct unusual driver name

please see if you can flash manually using stm32 cube programmer (NOT st-link utility)

[image]

mck1117, 2023-09-21:
>     * you have wrong (OLD) DFU driver and you would have to update DFU driver
> 
>     * or rusEFI console does not recognize that correct unusual driver name

Try connecting your H7 board to your PC. `DFU in FS Mode` is expected for an H7. It's a different bootloader implementation than the F4/F7.

beeedee, 2023-09-22:
Hello again,

When I tried with STM32CubeProgrammer, I can connect to the board and update it with a .hex file, but the onboard LEDs are dimly lit. I believe I need to perform a DFU update. Do I need an ST-Link to perform DFU updating on the STM32H7? Am I correct in my understanding?

I have some other questions bothering me as well. Does using the STM32H7 require us to enter any development processes?

beeedee, 2023-09-22:
Why we have both .dfu and .hex files? Are those files exchangable or are they serving different purposes, if they are different what is the purpose of those?

And also I saw bin files, what are those?

When loading the software to the board, is it enough to upload .hex file to the board?

Also as far as you know, is there anyone else who tried to use proteus with STM32H7?

By the way many questions I am asking right now is just to be sure, dont get mad because of the lack of knowledge :)