rusefi/rusefi #5582: fatal on tune with fuel sensor

rusefillc, 2023-09-25:
reset alphax
load GDI tune

actual result: fuel sensor is required

rusefillc, 2023-09-25:
[image]

mck1117, 2023-09-25:
https://github.com/fome-tech/fome-fw/commit/d98e9d1d547e6678b39eaec6988a5bc3edd3690c

rusefillc, 2023-09-25:
> [FOME-Tech/fome-fw@d98e9d1](https://github.com/FOME-Tech/fome-fw/commit/d98e9d1d547e6678b39eaec6988a5bc3edd3690c)

but does it hide fact that sensor is not initialized until reboot or is the sensor eventually initialized later?

mck1117, 2023-09-25:
It avoids any order-of-operations related error if you try to enable fuel pressure sensing without rebooting the ECU, or if you want to supply fuel pressure over CAN (Lua, etc)