rusefi/rusefi #5593: 6G72 Trigger Pattern

ElDominio, 2023-09-29:
**Is your feature request related to a problem? Please describe.**
I want the Mitsubishi 6G72 trigger pattern to be supported

**Would we need new configuration fields?**
An option under Trigger for Mitsubishi 6G72

**Describe the solution you'd like**
Add support for the pattern

[image]
factory trigger pattern analysis

[image]
bottom is crank (3 windows)  top is cam (4 windows)

rusefillc, 2023-11-08:
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#6g72-cam

ElDominio, 2023-11-30:
https://rusefi.com/online/view.php?log=1324

VVT reads, but no sync

ElDominio, 2023-11-30:
[6g72crankinglogsWOT.zip](https://github.com/rusefi/rusefi/files/13518421/6g72crankinglogsWOT.zip)

rusefillc, 2023-11-30:
[image]

rusefillc, 2023-12-07:
looks great

[image]

ElDominio, 2024-02-23:
This works! Why was it not closed?

closed 2024-02-23 by rusefillc